kevinakasam/BeltDrivenEnder3 #2: Z-plates have very thight holes for V-slot wheels

PerLangkjaer, 2020-12-04:
It seems that the holes for the wheels is about 1mm to close to eachother, 0.5mm on both sides of the belt.
At least compared to the CR-10 second Z-stepper plate you can buy of Aliexpress

kevinakasam, 2021-01-04:
> 
> 
> It seems that the holes for the wheels is about 1mm to close to eachother, 0.5mm on both sides of the belt.
> At least compared to the CR-10 second Z-stepper plate you can buy of Aliexpress

Thanks for that. Do you have issues with both sides? I updated the Left plate, because some are telling me that this does not fit. I made the gap 0.5mm wider. (And sorry for my late response)

PerLangkjaer, 2021-01-04:
Both sides are tight.
I resized it to 102% and it fitted a bit better.
And its ok with the reply,  X-mas, New Year and Covington...

kevinakasam, 2021-01-04:
okay maybe i should redesign the plates. I modified the Plates from the CAD that Creality supported. So the distance is 60,2mm between the holes. For me everything fits perfect.  I don't understand the problems with the Z-plates. For some they fit, for others they don't

kevinakasam, 2021-01-07:
So I spent some time and fexed the problem with the Plates. I made an mistake eith the drawing of the exentric nut, it was impossible to rotate them correctly, works now:)

kevinakasam, 2021-01-17:
So the files are updated, this issue is solved. Thanks for the hint:)

closed 2021-01-17 by kevinakasam

PerLangkjaer, 2021-01-18:
Hi, sorry for the late answer, ille give it a go when I have some spare time on the printer, but I'm sure it works fine with the updates you made.