kevinakasam/BeltDrivenEnder3 #14: LeftV3.3 stl

Viper7016, 2021-11-28:
I just printed the LeftV3.3 stl file and the teeth for the belt are all squished and not uniform. After comparing V3.3 to the previous versions, I see that all the older versions have a proper 2GT belt tooth, but 3.3 has a square 1mm tooth with 1mm spacin
[image]
g.

kevinakasam, 2021-11-28:
Hey, thats right. With the V3.3 I have changed the profile. The old profile does not fit the GT2 belt. However, the 1mm squares fit exactly to the belt. The edges are not rounded, so that these can "bite" into the curves of the belt and the belt is very well secured.
[image]

Viper7016, 2021-11-29:
Thank you for the quick reply!

I apologize for not comparing the belt. I only compared the STL. That v3.3 tooth pattern is the only part of the print that came out poorly. I have no idea why.

[image]

closed 2021-12-01 by Viper7016