kevinakasam/BeltDrivenEnder3 #15: 1_RightBottom.stl too close on gantry

ultrazauberer, 2021-11-29:
Hello,

thank you for the great modification. I could complete it yesterday and made three little observations:

1) the 1_RightBottom part (https://github.com/kevinakasam/BeltDrivenEnder3/blob/main/V3/1_RightBottom.stl) is on my Ender-3 too close to the gantry so the belt starts to rub on the 2040 profile. I fixed it for now by using two washers but 1mm - 1.5mm more clearance would be the best. So a simple increase of the strength ob the baseplate should solve this problem.
[image]

2) the pulleys on the top let my belt moving left and right an the clearance where the belt should move freely is pretty tight. You can see it in the picture.
[image]

3) the rod could be 4-5mm longer being in total 335mm. I let cut my rod before I assembled the mod and saw that the rod disappears in the 608ZZ bearings on the right and the left.

But all in all it's a really nice mod and I already did my first testprint and all went fine!

Thanks!

ultrazauberer, 2021-11-29:
I modified today the STL and reprinted the part. Now the part touching the 2040 profile is increased by 1,5mm.
[image]

I wonder why I am the only one with this issue.

weav2k4, 2022-01-01:
I concur the bottom idler pulley is too close to the vertical.  I will also be modifying the part to increase the clearance to get the belt parallel.  The left bottom and right bottom parts look identical to me so will be changing both. And you are correct as well that the rod needs to be 333mm minimum to be flush with the outer bearing faces.  Cut mine to 335 and have 1mm protruding the bearings.  Excellent mod though and hopefully will get it married to my gantry changes soon.

kevinakasam, 2022-01-10:
Hey, sorry, I forgot about your issue! I uploaded an update yesterday that made some changes to the belt path, I think your problem should be solved with that. Looking at the photos, it seems like you are using pure rubber belts? Are they reinforced (fiberglass for example)? If not, you should get new belts as soon as possible or you will have problems soon. Maybe the belts you are using are a little thicker than the reinforced belts.

ultrazauberer, 2022-01-10:
> Hey, sorry, I forgot about your issue! I uploaded an update yesterday that made some changes to the belt path, I think your problem should be solved with that. Looking at the photos, it seems like [...]

I ordered the belts from AliExpress (linked in the BOM). I think it looks fine:
[image]