kevinakasam/BeltDrivenEnder3 #20: No step file for linear rail parts

michelluccote, 2022-01-25:
I have been running this belt system over 6 months. I am now upgrading to linear rails but found one of the part does not fit correctly because my rails seem to have a slightly thicker car on them. I tried to modify the STL file for the left adapter but Fusion cannot generate an STL from the original STL. I import  and try to export it back and get an error. 

I downloaded the CAD files but this does not include any of the linear rail adapters for the Z axis. Can you please share these files?

kevinakasam, 2022-01-25:
You're right. I'm sorry, should be online now. Are you sure that your rail is to big? the Left parts slides a bit onto the rail like you can see on the photo. Maybe this helps :)
[image]
[image]

michelluccote, 2022-01-25:
Thanks for the fast reply. I'll take a look.

I bought some Twotrees MGN12N linear rails and they do seem thicker. Here is the link to them on Amazon.

https://www.amazon.com/gp/product/B07SN6MT6T/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&th=1

And a few pictures of the carriage aligned with my printed bracket. Everything else appears to align except for that thickness.
 
[image]
[image]

michelluccote, 2022-01-25:
I have downloaded the new STEP file and it has everything I needed. Great work by the way. I love this upgrade.

closed 2022-01-25 by michelluccote