kevinakasam/BeltDrivenEnder3 #21: CAD - MGN 12H Rail

HermanGaviriaMarin, 2022-02-04:
Lord Kevinakasam,

Your designs are flawless, they print perfectly fine, they work great. I can't express how glad I am to have found your mod for the ender series printers. 

I'm working on modifying my printer to accommodate for several extruders, moreover, I'm working with your MGN v3 design to use dual linear rails on the Z axis. Currently your cad model uses the v-slot wheels and there are no signs of the spacers you created for the linear rail variant of the mod, if it isn't too much to ask, would it be possible for you to post the CAD models to the other variants, more specifically one that has the spacers that replace the v-slot wheels?

Just to give you a little sneak peak here is a view of what I'm working on, to give you a little hint it involves the Octopus Board, 5 extruders, and a custom ultra silent fan shroud which uses noctua fans. Btw the motors up top are extruders and not the x steppers. Of course I was unable to change the model to reflect on your mod, not just yet, I was kinda hoping to get the spacers first (the step file from you) then I'd go in there and change things up 
[image]

[image]
This is pretty messy at the moment because I have not been able to take care of the wires but as soon as I have the extruders mounted where I need them then I'll do cable management. 

Thank you for your all your hard work, your mod really helped me.

kevinakasam, 2022-02-04:
Hey :D
Thats a huge project, 5 extruders will give you many colors thats cool! And I really like your Switchwire themed electronics box.
 So you mean the MGN Spacers arent included in the step file? Have you checked the last update of it? I thought I included everything, if not I'm very sorry and I will upload a full step file asap!
 One question though, are both z motors connected to the same rod?

HermanGaviriaMarin, 2022-02-05:
Grandmaster Kevinakasam,

Yea its a big project, I've been working on it for a while now. I will share all the step files with everyone once I'm done building it though to the truth is that this mod isn't really for everyone, only for those who have an octopus and plan on using it with a Bowden styled ender 3 series printer. However, the legs that I made I can most definitely say that many will come to like them as they are very strong and will be modular, as on now they only accept squash balls but I will make versions that use the old rubber feet as well as other feet that can be rigid mounted into wood with screws. Most of the things you see in my design are support free like your designs. 
[image]
 

I'm sure you didn't include the part because in the CAD you provide the mod is using the V-slot wheels which is totally understanble, if there weren't any linear rails than that specific piece didn't need to come up. I just happen to need it so that I can design some mods to accommodate for an extruder that happens to be located next to the secondary extruder mount you created. I could design my own spacer but I sorta want to keep your design as original as possible. It would be a huge help to have the cad with the all linear rail (dual linear rail) version on the mod. 
[image]
 

To answer your question about how I have those two motors connected to the same shaft while also being wired in parallel, the answer is very simple, I changed the polarity of one of the coils in the mirrored motor/transmission. Basically all I did was switch the A+ and A- to make the motor turn in the opposite direction, at the same time I left the other motor alone to spin in the normal direction. The result being that the motors will turn in the same direction despite being mirrored. I first performed this using the creality steppers which are wired with this sequence of coil sets: A+ X  B+ A- X  B-   <-- imagine that as the JST connector found on the stepper motor the x is just the pin in between that isn't used.  I'm sure you know all about this but I put all this info here just in case someone looks at this and has the same question
[image]

I have got some questions about your ender 3 afterburner design. More specifically the cable chains or drag chains. The drag chain that you have located in the Z axis (not the one on the X axis), where does that drag chain mount to in the bottom? I understand that it is  connected to the limit switch box that you designed but what about the bottom portion of that drag chains? I also want to add drag chains to my design but I will need two sets of Z drag chains because I will have those two extruders on the left side of the X axis. 
[image]

[image]

weav2k4, 2022-02-05:
I believe Kevin may be confused as I am about why you need two z steppers if they are joined?  You can just use a single stepper if not trying to do G34 with separate Z's.  Is it just so you have stronger Z action?  Very interesting and ambitious things going on here.  Curious to see how it all turns out for you!

HermanGaviriaMarin, 2022-02-05:
Weav2k4 I totally understand where you're coming from, my design is definitely not conventional in several ways but I'll try to point out my criteria and why I'm doing things like this. 

First, just as you briefly mentioned, yes it does have to do with the strength of the Z axis, the motors I'm using aren't per say extremely heavy, but there will be 4 of them in the X axis, that weight adds up, and though I strongly believe that one motor is likely to be enough I also like to play it save, thus having two motors. 

As to why I'm not running them individually. Well I really liked the idea of having independent motors, but I came to small conclusion that I have yet to have answered, I will create some illustrations for you so you can see where I'm coming from, please bare with me and my microsoft paint sketches. 

G34 was made to correct a skewed or non horizontal X axis on printers that have individual Z motors. The procedure is performed with a series of measurements taken from a bed probe such as a CR-Touch or BL-Touch, this probe measure the difference in height from across the bed in the horizontal distance of the X axis, it corrects this by adjusting the motors on each side of the Z axis individually to either raise or lower the gantry right? This is displayed below. 
[image]

But now lets explore what would happen if the bed is not in an ideal position. Assuming that the bed of the printer is perfectly leveled (and i don't mean tramming, I mean true leveling) then the X gantry will undoubtedly become perfectly horizontal and no side of the X gantry will be higher than the other (i.e the left side of the gantry is 2 mm higher than the right side of the gantry). But what if we perform a G34 and the bed is all sorts of ways skewed out of position? the G34 will perform what its being told to do, and that is make sure that the the left and right side of the X gantry are at the same height with relation to the bed thus the illustration below. 
[image]

Thus we are left in the same place where we started, the X axis is at an angle instead of being perfectly horizontal, yes it is not parallel to the bed but in the long term this will deform one or more components of the X gantry. 

I believe in having one rod and 2 motors because once you have set the same height on the gantry on both sides then they do not come undone for a really long time since you would have to forcefully make the belt slip out of the pulley's teeth and they will both move together as long as they are properly constrained, thus never letting the X axis become positioned at an angle. 

But yes, I do have two motors running in parallel with one motor running backwards because its located in a mirrored position. This is done because the gantry is becoming really heavy, this makes lifting the gantry and keeping it up there more easily.

weav2k4, 2022-02-05:
Totally understandable if needing the extra muscle if lifting a heavy gantry.  Happy modding!

HermanGaviriaMarin, 2022-02-05:
Thank you for the encouragement. By the way what is your opinion with regards to my little hypothesis/explanation about g34? Up to now no one has told me that I'm crazy for assuming that whole thing about G34 but I want to know what makers like you think.

weav2k4, 2022-02-05:
If one takes the time to make sure the gantry is parallel to the bed and stays that way then the need for two z's with g34 isn't necessary.  Many people with enders have the famous sagging gantry issues that can be tough to resolve if not lifting both sides.  That is why Kevin's creation is so great.  It's an elegant solution to the problem and has been flawless for me so far.  Other plus going belted is speedy z moves that a leadscrew just can't match.  The only thing a perfectly parallel setup doesn't resolve is a warped bed but that's what a BLTouch is there for.  If I could find a nice flat bed then could eliminate the bltouch but is nice to have that as a crutch to compensate for keeping the gantry trammed throughout the total bed area.  I try to keep things as simple as possible so a single z that lifts from both sides and does so mechanically is inherently less prone to issues.  I remember the days before bltouch was around.. sure is nice to have so many options for resolving issues now!

HermanGaviriaMarin, 2022-02-06:
Lord KevinAkasam, 

I hope you're well. I don't mean to rush you in any way, I know that you will be very busy with your exams and college. I just wanted to make sure you knew exactly what I was asking for, just to clear things up. When I meant for you to post the CAD to the other variants of the belted ender 3 mod I meant the versions that appear in your BOM. Below you will see what I mean.
[image]

Thank you for your work

kevinakasam, 2022-02-06:
Good morning,
sorry I forgot about my request. I have two exams tomorrow and am pretty stressed about it. I downloaded the CAD from Github and opened it in fzsuin. All MGN files are there, please look a little closer.
[image]

HermanGaviriaMarin, 2022-02-07:
Grandmaster Kevinakasam,

Yes you are indeed correct. The cad is complete and correct, I apologize for all the trouble. I must have been confused or must have enough experience with github, perhaps I ran into some sort of bug but I will explain why I think it wasn't working out for me. 

First, I hit the download code button, this let me download everything as zip file. After that I had extracted I directed myself to the stp file which I then opened in Autodesk Inventor, I found that the MGN files were not there, then I came here to ask for the cad. 
[image]
 
Later that day you responded to me and said you would upload the files, but since github has some sort of timestamp for all the files (shown below with red arrow) I assumed that the file had not been changed. 
[image]

That made me assume that nothing had changed. Today after reading your message I redownloaded the stp file, I opened it in Fusion and Inventor and it turns out that now it did in fact have the MGN files within the stp file. 

This is before
[image]

and this is after
[image]

Idk if its the timestamps of github which caused the confusion or not but either way thank you for all the help and I'm sorry If caused any trouble.

dwery, 2022-02-09:
@HermanGaviriaMarin I like the feet. Please let us know when you publish them!

HermanGaviriaMarin, 2022-02-09:
I'm so honored to hear that. I will publish them this week. I haven't published my designs yet since I want them to be perfect but if I wait I'll never get anywhere. I will share them soon friend. Thanks for the interest. All you need is 2 M5 screws and 2 M4 screws per foot. But I can also change them to be press fit, lmk which you'd prefer.

dwery, 2022-02-10:
I do like screws. No hurry, don't wait for perfection, but don't settle for anything less than very good either.

kevinakasam, 2022-02-10:
Good morning @HermanGaviriaMarin 
First of all, thank you for your detailed answers and the time you invest in your explanations. Therefore, I would like to answer you back a little more detailed. I'm currently in the middle of the exam phase and just have very little time and in the evening I'm usually tired and go to sleep early, which is why my answer took so long. So I want to give you some constructive feedback, please don't take this as simple criticism, they are just ideas :)

So your explanation about the G34 is correct, it aligns the axis to the bed and if that is crooked, the axis is crooked too. This is why you need a joint and a length compensation in your axis, otherwise you manage the leveling only by bending the plastic. That's why the Dual Z solution is not ideal at the moment. Regarding your project: Why did you mount both Z motors on the same rod? This makes a G34 and Tram Z physically impossible. No matter which of the two motors turns, it will always have the same effect. And since you have the two motors running in parallel, they can't rotate differently. That's the way I have it too, but I don't have them both connected to one rod, so the right and left side are controlled by the same motor signal, but by a Tram Z the axis can still align itself (e.g. the left side stops at its physical endpoint and skips steps until the right side also comes against its physical endpoint). This way the right and left side can be adjusted independently, with Tram Z but not with G34. For G34, the motors must also not be wired together, but each needs its own driver. With your setup you only have the same functions as with a single Z because they are both on the same rod. This might also be quite harmful for the electronics. The two motors may work against each other if they don't run exactly the same. That might demand a lot of current which might destroy your drivers in the long run.

I'm glad you found all parts in CAD, but I didn't change them, as you can see on the timestap. But I also sometimes get confused with all the CAD versions I have ;D

And I really like your mod with the grids at the bottom, it's just so it's not too high but it should give enough space to put the electronics inside. But I'm not so convinced about the feet. These squashballs make the printer much quieter but also much more unstable. at higher speeds the printer wobbles like a boat on the sea. Look at the squash feet of the Prusa bear (https://www.prusaprinters.org/prints/20292-low-profile-v-slot-squash-ball-feet-prusa-bear-log) they are just about ok, because the center of gravity of the printer is much lower compared to when the printer is above the squash balls. This makes it more stable, maybe that would be worth a thought? I once had a mixed solution on my printer, for this you cut the balls in half. The ball was filled by the foot and only the thick rubber damped the vibrations, but not the air in the ball. This makes the whole thing much more stable (similar to this, but the ball was filled in the middle by the print https://www.thingiverse.com/thing:3895933).

I can understand why you took the big fans for your part cooling, they are nice and quiet. But these "normal" fans are not really suitable for part cooling, because they don't have enough pressure and volume flow. The radial fans have the advantage that they are very strong, which is very good for part cooling, because we want to cool quickly and strongly if necessary. Only in this way the plastic can harden faster and the warm air can flow away. For materials that need little air, this may work, but I'm not sure if they can really keep up with one or two 5015.

My drag chain goes directly to the electronic case at the front. I just connected the chain the other way around in the midde to get the S shape. Like you can see here: 
[image]
[image]

I am excited about your results with the 5 colors, I also had a MMu2, but never used it and then sold it again. Now I have an Idex and I can use at least two colors. And do you want to mount all three or five motors on the X axis? Don't you think that will be a bit too heavy? Also, all the motors on top of the printer make it very top heavy which should give you a lot of vibrations. I would try to mount them on the side of your printer with two additional aluminum profiles. Like this maybe: 
[image]

So I think I have said something to all points. I am really curious to see how your printer looks like in the end. And I will definitely have a look at your electronics case design. I like that!

kevinakasam, 2022-02-27:
@HermanGaviriaMarin any Updates?

closed 2022-05-10 by kevinakasam