kevinakasam/BeltDrivenEnder3 #26: braces support addition

xjustbmw, 2022-02-22:
Are there any plans to add frame braces support on this version ?:)

Similar to this https://www.thingiverse.com/thing:3746653
I think this is the perfect Z axys combination for ender printers
My stock ender 3 printer had ~4mm deviation on top without the frame braces installed

Alfiegerner, 2022-02-23:
Was just about to ask something similar.

I'm upgrading to dual z from single motor.  Previously the two holes on for top left and top right on the standard belted version were just the tight size to fit the standard AliExpress bracing kits, but the dual z 3 bolt set up won't allow it.

xjustbmw, 2022-02-23:
We can surely mix the two parts together. But I think a new verion here for evrery one to enjoy would be nicer and have the printer way more rigid and square at the same time :)

kevinakasam, 2022-02-23:
Hello to both of you. Tbh I'm not the biggest fan of these support rods, but if you'd like we could add something similar to the mod. Should be pretty easy, especially with the normal top parts. And there should be a way to add it to the Dual Z setup as well

Alfiegerner, 2022-02-23:
Thanks @kevinakasam . Out of interest how come you don't like these braces? Just wondering if I've missed something.

Also, my print quality is so much better since doing this mod, really chuffed 🙏

xjustbmw, 2022-02-23:
Thanks again kevin for considering.

For me the quality really improved as well, but I also did dual Z rods, and direct drive with bmg/e3d all metal hotend at the same time. 
Took me a few hours to align them properly so it runs smoothly. but i would still like to move to belts in the future.

I think this braces only help if the gantry is not perfectly perpendicular from the factory, and if you print fast and tall objects. But surely they cannot hurt the printer . its way more rigid than before.

xjustbmw, 2022-02-23:
speaking of the future, any recommended better V slot wheels ?

kevinakasam, 2022-02-27:
could one of you cantact me via discord please? So we can talk about the additional support:)

xjustbmw, 2022-02-27:
sure. i just sent you a friend request. thanks

kevinakasam, 2022-05-10:
I've added a folder in V3_extras. I will close this now :)

closed 2022-05-10 by kevinakasam

xjustbmw, 2022-05-10:
thanks a lot for this, i am planning to print this soon
my printer works fine at the moment using the dual Z screws, but at the first error i will change it to belts :) already got all the required kit parts :)