kevinakasam/BeltDrivenEnder3 #28: Unequal belt path length for left and right belts in version 3.5

maxkueng, 2022-03-19:
Hi. I tried to upgrade to the V3.5 plates but I ran into an issue.

The distance from the top of the nut to the bottom of the belt clamp (see image below) is different for the left and right plate. The left distance is 19.405mm and on the right it's 20.629mm (blue dotted line in the image below).

This means that the required belt lengths are not the same for the left and right belts and therefore it's not possible to achieve equal belt tension on both sides without measuring the actual tension.

I took the same measurements from an older CAD (I don't know which version or how old) and there both distances are equal.

The method I use is to cut both belts to the same length and then tighten the tensioner screws with an equal number of turns. But for that the belt paths on both sides must be exactly the same.

The measurements below were taken from BeltDriven_Ender3_V3_25012022.step and I've also been able to measure the difference in the actual prints.

[image]

maxkueng, 2022-03-19:
Here's a Gif of how I fixed the RightV3.3 plate so that the belt paths are equal. I shortened the belt path on the right plate so that it's the same as for the left plate. 

I don't have Fusion 360 so I did it in Onshape. Here's the link:  
https://cad.onshape.com/documents/530c50c0feb35506422f7b54/w/d6230fb67fb3cda85e9c2f90/e/6c6611e621e6919f323474aa?renderMode=0&uiState=623618cd12c52217a3e5bc8e

[image]

kevinakasam, 2022-03-20:
Hello!
also thank you very much for this hint! I am wondering how this error could happen since I mirrored the plates, but I will fix this as soon as possible. I can't believe you caught this problem^^

aidbish, 2022-05-06:
@kevinakasam was the files updates or is it best to use @maxkueng version above to resolve this.

maxkueng, 2022-05-08:
@aidbish In case you need it, here's a Zip with STL and STEP versions of the fixed part. I couldn't upload the STL directly because GitHub doesn't support it.

[BeltDrivenEnder3 - RightV3.3-fixed.zip](https://github.com/kevinakasam/BeltDrivenEnder3/files/8647390/BeltDrivenEnder3.-.RightV3.3-fixed.zip)

aidbish, 2022-05-11:
@maxkueng Thanks mate, appreciate it

maxkueng, 2022-12-03:
@kevinakasam It looks like this has been fixed in V3.6. Can you confirm?
While I haven't printed the new version, I did check the new 3.6 CAD and it looks like the belt paths are now the same length.

So maybe you can close this issue.