kevinakasam/BeltDrivenEnder3 #30: Left/Right Top AB for Rotated transmission

ljruiten, 2022-03-24:
So please correct me if I'm wrong, but I'm trying to find in the documentation how I could have known that the parts I printed were not compatible.
From this Git I took the latest version, and printed the transmission body with bearing and rotated version. Then according to BOM I printed the left/right top A/B, but they don't align with the rotated transmission body for the rod. Looking at the STL files they do align with the unrotated version though. The only bearing holders that seem to align with the rotated transmission body are the TransmissionBody_Left.stl and TransmissionBody_Right.stl in the root directory. This seems like there's a line missing in docs/BOM maybe?

ljruiten, 2022-03-24:
See this picture this is:
- BeltDrivenEnder3\V3\Voronparts\1_transmission_rotated_bearing.stl
- BeltDrivenEnder3\V3\Voronparts\1_RightTop_A.stl

Perhaps what I didnt get is that the rotated version IS the ultra low version, but if you look at attached picture the two holes for the bearing don't line up if you put them on top of the frame.

The what you call "ultra low version" does seem to be compatible with the rotated version. Just wasn't clear to me that those two belonged together.

[image]

kevinakasam, 2022-03-24:
Or this top part for your transmission: https://github.com/kevinakasam/BeltDrivenEnder3/blob/main/V3_Extras/UltraLowHeight(Ikea_Lack_Enclosure)/Single_Z/LeftTopA.stl

kevinakasam, 2022-03-24:
So this are the normal files (not the ultra low version)[image]

ljruiten, 2022-03-24:
Ok, well I think I rather reprint the tranmission than the top parts. Just thought the rotated version looked a little cooler. I get it now.

Have to say: it is a little confusing to find the right set of files.

ljruiten, 2022-03-24:
Anyway I'll close this issue. Thanks for the support

closed 2022-03-24 by ljruiten

kevinakasam, 2022-03-24:
You just have to print the files from the V3 folder and pick a transmission body you want (rotated or unrotated/withbearing or without) I'm sorry for the stls in the root folder, I guess that was the issue