kevinakasam/BeltDrivenEnder3 #38: Klipper Rotation Distance OR ESteps

MahmoudHassib, 2022-05-10:
Hello, Perfect Mod 
But what is the Prober Rotation Distance after upgrading to this mod since we are changing from normal T8 Leadscrew

igor-arkhipov, 2022-07-21:
Same as before, rotation_distance = 8

Gear and pulley comes from Voron 2,  here is example config – https://github.com/Klipper3d/klipper/blob/master/config/kit-voron2-250mm.cfg#L86

kevinakasam, 2022-07-26:
Hey,
I now it's been some time but I haven't checked the issues tab for a while, sorry. The mod works with the stock settings, but for Klipper you could add the gear ratio, like this:

`rotation_distance: 40
gear_ratio: 80:16            
microsteps: 16
full_steps_per_rotation: 200    #200 for 1.8 degree, 400 for 0.9 degree`

MahmoudHassib, 2022-07-26:
@igor-arkhipov @kevinakasam 
thanks ,i can assure it works as same as in marlin with out any change if you are using normal T8 lead screw , but mainly i was concerned in regards to Klipper and the gear ratio  value with the rotation distance .
@kevinakasam  it would be nice if you can add  Klipper Rotation Distance and Estep's in the manual

MahmoudHassib, 2022-07-26:
@igor-arkhipov @kevinakasam 
thanks ,i can assure it works as same as in marlin with out any change if you are using normal T8 lead screw , but mainly i was concerned in regards to Klipper and the gear ratio  value with the rotation distance .
@kevinakasam  it would be nice if you can add  Klipper Rotation Distance OR Estep's