kevinakasam/BeltDrivenEnder3 #45: Transmission_UL

phanthuat, 2022-08-07:
I'm on version 3.5 and am planning to switch back to v3.6 UltraLow.
Can you update the file Transmission_UL_V3.6 for the left side? because if using the right file, the stepper motor cable will have to be much longer? thank you very much

kevinakasam, 2022-08-07:
> I'm on version 3.5 and am planning to switch back to v3.6 UltraLow.
> 
> Can you update the file Transmission_UL_V3.6 for the left side? because if using the right file, the stepper motor cable will have to be much longer? thank you very much

Hey,
You can use the left side file from the dual z version. It's just mirrored. This way you can also mirror the top parts for the other side (just do it in slicer). If that's too complicated join my discord please and give me a ping there, this way it's easier for me to keep up with all the requests :)

phanthuat, 2022-08-07:
oh, it's so simple I couldn't think of it, thank you so much

closed 2022-08-07 by phanthuat