kevinakasam/BeltDrivenEnder3 #64: Ender3 v2 neo uses different extrusions

kandrejevs, 2023-01-25:
An issue I have encountered is that the new "neo" update to ender series uses different extrusions with 1 side flat. 
It is still fine in my opinion since support is held with 2 screws and motor mount with 3 if you rotate extrusion 180 degrees. 
[image]

shuff57, 2023-02-09:
@kandrejevs I modified the files to work on an elegoo neptune 2 which appears to have the same style of extrusions. If you're interested, I can send/attach the .stl files of the modded parts along with some pictures?

jonathanysp, 2024-02-28:
@shuff57 This is an old issue, but i'm interested in your modified stl files that would work with new style of extrusions.

Sartium, 2024-03-01:
@jonathanysp I just upgraded my V2 Neo to the Z belt and I found it easier to just buy a new 2020 extrusion and cut/drill it. Took 10 minutes if you have the tools.

shuff57, 2024-03-02:
@jonathanysp I can't attach .stl or .3mf files so I'm adding google drive link for you get the files. I'll try to add some pictures in there too.

[Elegoo Neptune 2 - Belted Z Mods](https://drive.google.com/drive/folders/1ToEX0puyeLmGaZNeT88EKgmVN3ZtdQ0N?usp=sharing)