kevinakasam/BeltDrivenEnder3 #78: Add recommendation for PC wheels

aamott, 2024-09-28:
Didn't know where to put this, so sorry for that, but it made such a difference with my belted z setup that I had to share. 

After something like two years of using this mod, I decided to try PC wheels on the z axis. I'm on the old a4988 drivers, but it looks better than any printer I've ever used before. It's an insane difference switching between PC and Delrin wheels. $3 for the whole upgrade (used the old bearings) but it looks like the work of linear rails. I'd like to recommend a note on PC wheels in the readme because of this.

Jonathan-Ott, 2024-09-29:
Adam, do you have a link to the ones you used?

aamott, 2024-09-30:
Yep! I bought these: [Polycarbonate Xtreme v Mini wheel for Openbuilds](https://www.aliexpress.us/item/3256805539482478.html)

Amazon has some too, though these have bearings preinstalled, no idea of the quality: [Clear Polycarbonate Wheel](https://amzn.to/4ep9UoM)

I had no chance of getting the bearings back out of the polycarbonate wheels without cutting them. Getting them out of the delrin was no problem though once I realized I couldn't push from one side to get them both out. That little plastic lip in the center is obvious in the clear wheels. 

Also, in my research, only one person did real tests on the X/Y axes with PC wheels vs delrin and found their first few hours made noticeable dust, just like delrin, but after they wore in it was barely noticeable (again, like delrin). I'll test X and Y later too, since now the rest of the printer is quality enough to see any difference, but may be a little.